esp-rs/embuild #1: bindgen.rs - Windows build -  failed to resolve: could not find `unix` in `os`

georgik, 2021-06-28:
The build on Windows fails with:
```
   Compiling cargo-pio v0.8.0 (https://github.com/ivmarkov/cargo-pio#d9e15e94)
error[E0433]: failed to resolve: could not find `unix` in `os`
 --> C:\Users\JurajMichálek\.cargo\git\checkouts\cargo-pio-1b39afa97d59a391\d9e15e9\src\bindgen.rs:1:36
  |
1 | use std::{env, ffi::OsStr, fs, os::unix::prelude::OsStrExt, path::{Path, PathBuf}, process::Command};
  |                                    ^^^^ could not find `unix` in `os`

error[E0599]: no function or associated item named `from_bytes` found for struct `OsStr` in the current scope
   --> C:\Users\JurajMichálek\.cargo\git\checkouts\cargo-pio-1b39afa97d59a391\d9e15e9\src\bindgen.rs:115:50
    |
115 |         Ok(fs::canonicalize(PathBuf::from(OsStr::from_bytes(&output.stdout)).canonicalize()?)?)
    |                                                  ^^^^^^^^^^ function or associated item not found in `OsStr`

error: aborting due to 2 previous errors
```

The first error is cause by using `os::unix`. This can be solved:
```
#[cfg(unix)]
use std::{os::unix::prelude::OsStrExt}

#[cfg(windows)]
use std::{os::windows::prelude::OsStrExt}
```

ivmarkov, 2021-06-28:
I've fixed ti so that it should compile now on Windows, but have not tested it yet. If you can give it a try, that would be greatly appreciated.

ivmarkov, 2021-06-28:
(Just pull latest rust-esp32-std-hello and try to rebuild - it will re-fetch cargo-pio.)

georgik, 2021-06-29:
Both errors were solved. Thank you

closed 2021-06-29 by georgik