esp-rs/embuild #3: Link failure (Rust 1.54) esp32s2

MabezDev, 2021-08-02:
Building esp std hello from this tree: https://github.com/ivmarkov/rust-esp32-std-hello/tree/07b7fb9a8f11b21c709c5277377f8f03439e27f1, the only modifications being the SSID, Pass & target in config.toml.

```rust
    Updating crates.io index
    Updating git repository `https://github.com/ivmarkov/libc.git`
   Compiling rust-esp32-std-hello v0.14.2 (/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello)
error: linking with `cargo-pio-link` failed: exit status: 1
  |
  = note: "cargo-pio-link" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.17pamn1c5ias0as1.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.1b6lc6ms2f18f3ea.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.1dbj2xzkvf86f1gy.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.1itjfh7gmsk7xbh7.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.1ost5lt6ejffkz3y.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.1wphx7uhjupoe75r.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.221k79kujycpmgj6.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.283wtpvs0uehl2c.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.2at60ydjj69y9th3.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.2drchrzhlijfafmp.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.2if95hp17n23m85h.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.2igywquu67sxhx95.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.2r5r5y2vkahdqjmw.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.2u5p27ut4w46gf49.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.2x6ptb08qvhyvlvq.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.2xbykl9hs1hkgtrl.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.2zd836dw9z45hg3d.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.3338v3rlvl7hu03z.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.3g0hsp8aid484cm.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.3j2kcte3d64oljz1.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.3mt04fofnvi9tgi8.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.3vovshvqxgcvug9t.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.42d0z9ydp5nn4x0s.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.47hfz4u9qf4ib9c0.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.4d6opkzn2dp5svql.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.4g2lv447yyb8wel.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.4gr1svn1fpg70x8q.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.4lxo8wom6ymo0csf.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.4pujlbzwxie4uupq.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.4wo57lqcos7za6n4.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.4ykxpynyg5x1n813.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.50tgedq46pafn9tl.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.54ky6s68jllwfl7j.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.55jx9vctbmo49t16.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.5a5tk4urairkf39e.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.5cug23beti6uewug.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.5kp51rr6ykdb901.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.dctido0t0eyejci.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.sdaxg9ew1txwm6f.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.sg9fjlb2rakwagu.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.zvg6f8qvsi9pwyk.rcgu.o" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/rust_esp32_std_hello-93ff9ee32e6fc5bc.5etwit9xqrlrlq2r.rcgu.o" "-Wl,--as-needed" "-L" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps" "-L" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/debug/deps" "-L" "/home/mabez/development/rust/xtensa/rust-xtensa-dev/build/x86_64-unknown-linux-gnu/stage2/lib/rustlib/xtensa-esp32s2-espidf/lib" "-Wl,-Bstatic" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/libst7789-ac9dda3d8b99bd12.rlib" "/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/target/xtensa-esp32s2-espidf/debug/deps/libh [... cut ...] =/builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.4.0 (crosstool-NG esp-2020r3)
```
error status 1 doesn't really tell us a lot unfortunately. Any ideas?

ivmarkov, 2021-08-02:
Yep. Caused by the removal of my shameful "redirects" of `pthread_rwlock_t` to `pthread_mutex_t` as described [here](https://github.com/rust-lang/libc/pull/2310#issuecomment-891117457).

I've ["fixed it"](https://github.com/esp-rs/esp-idf-sys/blob/master/src/pthread_rwlock.rs) a couple of hours ago, but the truth is, we need `pthread_rwlock_t` implementation (either there or as a patch against the ESP-IDF) ASAP. I really do not want to return to the old Rust RwLock implementation in my libStd patches, as that was really difficult to maintain and increased the patch size significantly. The libStd patches (IMO) are much smaller & cleaner now, due to that and the `lwip_` proxies removal.
 
What you need to do:
- `cargo install cargo-pio` to get the new cargo-link proxy, which **does** report the linker errors coming from the GCC tollchain linker :). By the way, the patchset against libc will allow us to get rid of this proxy. That is, provided that the ESP-IDF GCC toolchain is on your `$PATH`
- In the root of the rust compiler fork: `cargo update -p libc` (not sure this is absolutely needed, with my force-pushes to libc though)
- in the root of `rust-esp32-std-hello`: `cargo update`

MabezDev, 2021-08-02:
Hmmm doing all that hasn't helped unfortunately.

I forced installed pio from git: `cargo install cargo-pio --git https://github.com/ivmarkov/cargo-pio --force`

Cargo.lock in rust compiler:
```toml
[[package]]
name = "libc"
version = "0.2.98"
source = "git+https://github.com/ivmarkov/libc.git#85210e6b2e226216431f80f5a0b3ebb329f98adb"
dependencies = [
 "rustc-std-workspace-core",
]
```

```bash
rust-esp32-std-hello git:(main) ✗ cargo update
    Updating git repository `https://github.com/ivmarkov/libc.git`
    Updating crates.io index
    Updating git repository `https://github.com/ivmarkov/ili9341-rs`
    Updating esp-idf-svc v0.16.5 -> v0.16.6
    Updating esp-idf-sys v0.16.3 -> v0.16.4
```

Also just for transparency, some env info:

```
which clang
/home/mabez/development/rust/xtensa/llvm-project/build/bin/clang
```

I can't seem to capture the linker output from https://github.com/ivmarkov/cargo-pio/blob/01c8113428708d97906f60d4046a3b0421c3cecd/src/link.rs#L105-L112 

All I see in the terminal is the status line: `Error: Linker /home/mabez/.platformio/packages/toolchain-xtensa32s2/bin/xtensa-esp32s2-elf-gcc failed: exit status: exit status: 1`

ivmarkov, 2021-08-03:
I'm really sorry - I've actually **forgotten** to push and re-publish cargo-pio!
If you re-install it (crates.io is also ok) you should be able to see it.

(These debug statements  needed `export CARGO_PIO_LOG=debug` and then `cargo build --verbose`if I'm not mistaken.)
The fix is making sure that the stderr redirect of the proxied linker is displayed regardless of the log level, as long as there is a failure.

ivmarkov, 2021-08-03:
Guess, I was sleeping yesterday. Also `rust-esp32-std-hello` is now pushed to git, with the correct changes.

MabezDev, 2021-08-03:
Haha no worries! By updating pio link, it showed me pio wasn't building the ULP loader & runner - after that a quickly figured out my cargo was too old. I now have it running on my esp32s2 here!

Thanks for the help!

closed 2021-08-03 by MabezDev

ivmarkov, 2021-08-03:
The ULP thing has two annoyances:
- Smallish: you need to extend the max length HTTP headers with `cargo pio espidf menuconfig`. We should probably then look at the generated `sdkconfig.debug` and manually move the HTTP headers param into `sdkconfig.default.esp32s3-ulp-example`
- Largish: the main CPU does NOT go to sleep as expected, but immediately resets when I tell it to go to sleep. I _suspect_ the ULP bootstrapping code somehow generates a trap (maybe I've still missed an invalid instruction or something in the bootstrapping code I copied from the riscv crate), and because of that, the main CPU is awoken immediately. In the meantime, the ULP program is executed though, the LED (my voltmetter actually) blinks, etc. I'm a bit at a loss what I might be missing here

MabezDev, 2021-08-03:
> Largish: the main CPU does NOT go to sleep as expected, but immediately resets when I tell it to go to sleep. I suspect the ULP bootstrapping code somehow generates a trap (maybe I've still missed [...]

Commenting out the launching of the ULP app and esp32s2 goes into a deep sleep and stays there for about a minute - it must be ulp related.

I added some code to print the sleep wake up reason and I got the following:

`rust_esp32_std_hello: Sleep wake up reason: 0x0B`

Looking at the IDF source:

```
typedef enum {
    ESP_SLEEP_WAKEUP_UNDEFINED,    //!< In case of deep sleep, reset was not caused by exit from deep sleep
    ESP_SLEEP_WAKEUP_ALL,          //!< Not a wakeup cause, used to disable all wakeup sources with esp_sleep_disable_wakeup_source
    ESP_SLEEP_WAKEUP_EXT0,         //!< Wakeup caused by external signal using RTC_IO
    ESP_SLEEP_WAKEUP_EXT1,         //!< Wakeup caused by external signal using RTC_CNTL
    ESP_SLEEP_WAKEUP_TIMER,        //!< Wakeup caused by timer
    ESP_SLEEP_WAKEUP_TOUCHPAD,     //!< Wakeup caused by touchpad
    ESP_SLEEP_WAKEUP_ULP,          //!< Wakeup caused by ULP program
    ESP_SLEEP_WAKEUP_GPIO,         //!< Wakeup caused by GPIO (light sleep only)
    ESP_SLEEP_WAKEUP_UART,         //!< Wakeup caused by UART (light sleep only)
    ESP_SLEEP_WAKEUP_WIFI,              //!< Wakeup caused by WIFI (light sleep only)
    ESP_SLEEP_WAKEUP_COCPU,             //!< Wakeup caused by COCPU int
    ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG,   //!< Wakeup caused by COCPU crash
    ESP_SLEEP_WAKEUP_BT,           //!< Wakeup caused by BT (light sleep only)
} esp_sleep_source_t;
```

We can in fact see the wake up cause is `ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG`.

However like, I put my multimeter on pin 5 and can see it blinking too - very strange behavior.

ivmarkov, 2021-08-03:
It blinks because even though the ULP cpu gets a trap, the trap is ["processed"](https://github.com/esp-rs/esp-idf-hal/blob/master/ulp/ulp_start.S#L57) with a `ret` riscv instruction and I guess therefore the ULP can continue with the bootstrapping procedure. Moreover, the main CPU and the ULP are completely disconnected in terms of execution once the ULP program is started - the main CPU might reset, reboot, whatever - the ULP does not care and continues chugging along.