esp-rs/embuild #4: Error: matches multiple boards

LucianoBestia, 2021-08-08:
When I run this command:
`$ cargo pio new -b sipeed-longan-nano my_blink2`  
I got this error:  
`Error: Configured board 'sipeed-longan-nano' matches multiple boards in PIO: [sipeed-longan-nano, sipeed-longan-nano]`  

How can I correct this?
I asked PlatformIO community if it is normal to have multiple boards with the same ID and they answer that is normal: "A board is only unique per platform." <https://community.platformio.org/t/multiple-board-id-sipeed-longan-nano/22827>
So I added the plaform:  
`$ cargo pio new -b sipeed-longan-nano -p GD32V my_blink2`  
But I got the same error.
I there something I missed and need to add?

ivmarkov, 2021-08-08:
Nothing missing really, it is just that the current resolution algorithm *assumes* that the board name is unique across platforms. Let me try to fix it. Once/if fixed, your command `cargo pio new -b sipeed-longan-nano -p GD32V my_blink2` should be working.

ivmarkov, 2021-08-08:
OK, so I've published a new version of `cargo-pio` so you can try it out by re-issuing:
- `cargo install cargo-pio`

The following command line now works:
- `cargo pio new -b sipeed-longan-nano -p gd32v my_blink2` (please note the lowercase name of the platform; matching is case sensitive)

... however, I cannot test the build of the resulting crate, because of [this issue in PlatformIO](https://github.com/sipeed/platform-gd32v/issues/6).

Hope that helps.

ivmarkov, 2021-08-08:
By the way... this board looks like a pretty simple thing - no WiFi, BLE or anything like that. And only 32KB RAM / 128KB flash.

So if you really want to code in Rust for it, I don't see the need to go the Cargo->PIO route. Why not just using this [crate](https://github.com/riscv-rust/seedstudio-gd32v), and forget about PlatformIO?

closed 2021-08-08 by ivmarkov

LucianoBestia, 2021-08-09:
Great! Thank you. 
Now it works flawlessly. Yes, Longan Nano is super cute and it has a Risc-V processor. I am a "fan" of rust and Risc-V and wanted to play with this combination.
Thank you for the tip for seedstudio-gd32v. I will give it a try.
I wrote some instructions how to setup a development environment for Longan Nano in Win10 + WSL2 Debian for the rust language:
https://github.com/LucianoBestia/longan_nano_rust_wsl2_platformio_setup
I needed to solve some workarounds, but now it works. 
I read lately that the ESP32-C will have a Risc-V processor with Wi-Fi and Bluetooth 5. Sounds like a good future.

ivmarkov, 2021-08-09:
> Great! Thank you.
> Now it works flawlessly. Yes, Longan Nano is super cute and it has a Risc-V processor. I am a "fan" of rust and Risc-V and wanted to play with this combination.
> Thank you for the tip for seedstudio-gd32v. I will give it a try.

The `longan-nano` crate actually looks like even more appropriate for your board, so I guess you should keep using that.

> I wrote some instructions how to setup a development environment for Longan Nano in Win10 + WSL2 Debian for the rust language:
> https://github.com/LucianoBestia/longan_nano_rust_wsl2_platformio_setup

That's great and I'll put `edition = "2018"` by default in the autogenerated `lib.rs` by cargo-pio. However my point is: you **don't need** `cargo-pio` actually?
Your project relies on a pure-Rust crate (`longan-nano`) and it is not interfacing with any Vendor SDK and/or C APIs. So why bother with PlatformIO and `cargo-pio` at all?

Just turn your demo crate from "static lib" to a binary crate one, say `!#[cfg(no_main)]` and you should be good to go. The build will generate a regular `.elf` file instead of the `.hex` one you need for flashing, but I'm sure there are pure-Rust utilities, which can do `.elf` to `.hex` and then do the flashing for your board.  

> I read lately that the ESP32-C will have a Risc-V processor with Wi-Fi and Bluetooth 5. Sounds like a good future.

Yes. That's the `ESP32-C3` chip which is on my desk as we speak. Now, the `ESP32-C3` is a whole different beast compared to your board - 10x more RAM, 40x more flash, and very difficult to write the WiFi & BLE layer from scratch in Rust, so you have to interface with the ESP-IDF SDK APIs. That's a chip where `cargo-pio` (either in its cargо-first variant or PIO-first) is really useful.