esp-rs/embuild #6: cargo-pio eats linker args

MabezDev, 2021-08-13:
```toml
rustflags = [
  "-C", "link-arg=-Wl,-Map=build.map",
]
```
Does not produce a map file.

I also tried a hard coded path:

```toml
rustflags = [
  "-C", "link-arg=-Wl,-Map=/home/mabez/development/rust/embedded/experiments/rust-esp32-std-hello/build.map",
]
```

Which also did not work

N3xed, 2021-08-22:
I think this is because there is already an argument to generate a map in the linker arguments. The generated map should be located at `target/<target>/<build or release>/build/esp-idf-sys-<hash>/out/esp-idf/.pio/build/debug/esp-idf.map`. 

This is a current limitation as we just forward all linker arguments from the `esp-idf` to rust. Until we do this more sophisticated (which shouldn't be long), just use that file as a workaround.

MabezDev, 2021-08-22:
Ah! Great spot! 

I remember recursively searching for build.map, thinking it might have been generated within the "mini idf project", but I don't think I searched for `*.map`.

Lets keep this open for now (I'll rename the issue) until we have a better way of dealing with this.

MabezDev, 2021-11-04:
I tried this again, and this now seems to work! By default the map file is place relative the esp-idf-sys directory, but atleast now explicit paths now work.

Closing.

closed 2021-11-04 by MabezDev