esp-rs/embuild #19: Cannot create a cargo pio project with the esp32

Vandell63, 2021-09-23:
Hi, and first thx for all the hard work you've done.
I'm just starting to try the `cargo-pio` tool.
So i've executed this first command: 
`cargo pio new -b esp32dv cargo-pio-test'
the platform.ini is : 
```
[platformio]
default_envs = debug

[env]
extra_scripts = pre:platformio.git.py, pre:platformio.patch.py, platformio.cargo.py
board = esp32dev
platform = espressif32
framework = arduino
rust_lib = platformio
rust_target = xtensa-esp32-espidf

[env:debug]
build_type = debug

[env:release]
build_type = release
```
After running the command: 
`cargo pio build`
with both `arduino` & `espidf` framework
I got error : 
```
Building .pio/build/debug/bootloader.bin
esptool.py v3.1
Merged 1 ELF section
__run_cargo([".pio/build/debug/firmware.elf"], [".pio/build/debug/src/dummy.o"])
*** [.pio/build/debug/firmware.elf] TypeError : str expected, not tuple
Traceback (most recent call last):
  File "/home/vandell/.platformio/packages/tool-scons/scons-local-4.2.0/SCons/Action.py", line 1279, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "/home/vandell/Dev/Rust/Iot/Libs/platformio-test/esp32dev/platformio.cargo.py", line 54, in __run_cargo
    env["ENV"]["CARGO_PIO_BUILD_PROJECT_DIR"] = env.subst("$PROJECT_DIR"),
  File "/usr/lib/python3.9/os.py", line 684, in __setitem__
    value = self.encodevalue(value)
  File "/usr/lib/python3.9/os.py", line 756, in encode
    raise TypeError("str expected, not %s" % type(value).__name__)
TypeError: str expected, not tuple
==================================== [FAILED] Took 17.45 seconds ====================================

Environment    Status    Duration
-------------  --------  ------------
debug          FAILED    00:00:17.452
=============================== 1 failed, 0 succeeded in 00:00:17.452 ===============================
```
I tried to change the framework to the `espidf`
and exactly the same error.

`rustc 1.57.0-nightly (9dd4ce80f 2021-09-17)`
`cargo 1.57.0-nightly (33ee5f82e 2021-09-17)`
__________
I try using the esp32 directly from creating the project from pio.
And it works as expected.
Any idea ?

ivmarkov, 2021-09-23:
I'll check what is going on and update you later.

However: if you are targeting the ESP32, with the ESP-IDF platform + **Rust**, I strongly suggest you take the complete opposite of cargo-pio, which is the so called "Cargo-first" build. A mix of PlatformIO and Rust where PlatformIO is in charge of the overall build is only useful if you plan to create a mixed C/Rust codebase, and interoperate between the two. Possible, but difficult.

Use this crate as a start: https://github.com/ivmarkov/rust-esp32-std-hello

Vandell63, 2021-09-23:
Thx for your response.
So yes, i succeed to build with the `rust-esp32-std-hello`, and it works pretty well. Good job there.
Im just starting in the embedded world, so i give a shot over here.
The reason why i give a shot here are :
- I have an esp32-cam , so i was searching for, 'as you said' a mixed between 'C' & 'Rust' to get the esp32-cam driver over here https://github.com/espressif/esp32-camera. Maybe it is not the correct way to do it. Maybe i should give a try with the `bindgen` to get the job done between Rust & C here.
- I was also searching about the bluetooth & the esp32Now communication but i did not find any starting point. Maybe you could lead me into the correct way.
Thx in advance.

ivmarkov, 2021-09-23:
So first of all in the meantime I've fixed the original issue (sorry!) and PlatformIO-driven builds should work again. Just re-install with `cargo install cargo-pio`.

As for your particular case... you are in for quite a journey - if you continue using the cargo-first build I just suggested to you!

It is possible, but I actually need to think how this is best done, because so far - honestly - cargo-first was always about just using the "standard" ESP-IDF. For your case, we might have to "extend" the build somehow so that you have a way to "point" the build to the additional C components you would like to build as part of building ESP-IDF.

(With the pio-first project you started originally that's much easier, as you can just "paste" the esp-cam driver in the pio project directory.)

Regardless which way you decide to take, I won't bother initially with "bindgen", as this is another complication, and the cam driver interface is small enough so that you can manually create the unsafe Rust `extern "C"` functions of the driver.

I'll follow-up if I get a good idea for cargo-first.

Vandell63, 2021-09-23:
Thx ! 
I succeed to build with the `arduino` framework. 
But it seems to have an issue with `espidf` framework. 
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.02s
/home/vandell/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: target/xtensa-esp32-espidf/debug/libesp32_dev.a(compiler_builtins-8567a693cfe2fca1.compiler_builtins.b32yuzpz-cgu.14.rcgu.o): in function `__moddi3':
/home/vandell/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.49/src/macros.rs:285: multiple definition of `__moddi3'; /home/vandell/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/no-rtti/libgcc.a(_moddi3.o):/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libgcc/libgcc2.c:1256: first defined here
/home/vandell/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: target/xtensa-esp32-espidf/debug/libesp32_dev.a(compiler_builtins-8567a693cfe2fca1.compiler_builtins.b32yuzpz-cgu.15.rcgu.o): in function `__divdi3':
/home/vandell/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.49/src/macros.rs:285: multiple definition of `__divdi3'; /home/vandell/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/no-rtti/libgcc.a(_divdi3.o):/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libgcc/libgcc2.c:1232: first defined here
/home/vandell/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: target/xtensa-esp32-espidf/debug/libesp32_dev.a(compiler_builtins-8567a693cfe2fca1.compiler_builtins.b32yuzpz-cgu.15.rcgu.o): in function `__udivdi3':
/home/vandell/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.49/src/macros.rs:285: multiple definition of `__udivdi3'; /home/vandell/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/no-rtti/libgcc.a(_udivdi3.o):/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libgcc/libgcc2.c:1319: first defined here
/home/vandell/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: target/xtensa-esp32-espidf/debug/libesp32_dev.a(compiler_builtins-8567a693cfe2fca1.compiler_builtins.b32yuzpz-cgu.15.rcgu.o): in function `__umoddi3':
/home/vandell/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.49/src/macros.rs:285: multiple definition of `__umoddi3'; /home/vandell/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/no-rtti/libgcc.a(_umoddi3.o):/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libgcc/libgcc2.c:1307: first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/debug/firmware.elf] Error 1
================================================ [FAILED] Took 17.57 seconds ================================================

Environment    Status    Duration
-------------  --------  ------------
debug          FAILED    00:00:17.566
=========================================== 1 failed, 0 succeeded in 00:00:17.566 ===========================================
```

Honestly, i spent several weeks on the esp rust libraries to try understand all the things going on there.
From starting with the `esp32-hal` or with the `esp32-std-example` for the http, wifi etc.. features, and now for my case as you suggest, for the `platformio rust` specially (in my case) if have a need of importing external "C" methods in.

I think it could be really really nice if you could give an example with the `platformio rust` with a part of the `esp32-idf` features and maybe a random external "C" example and make them dancing together for a nice `elf` result. Im pretty sure that you will bring more people over here.
A nice 'firework'. 
I did not found any better rust embedded integration experiment than yours so again thx for this job.
At least any suggestion or good starting point with rust for `bluetooth` & `espNow` communication ?.

ivmarkov, 2021-09-24:
> But it seems to have an issue with `espidf` framework.

This is now fixed:
* Please update with `cargo install cargo-pio`
* Note that the fix requires that you re-create your project from scratch. But there is a better way, read on.

> I think it could be really really nice if you could give an example with the `platformio rust` with a part of the `esp32-idf` features and maybe a random external "C" example and make them dancing together for a nice `elf` result. Im pretty sure that you will bring more people over here.

There's now a PIO-first demo project specifically for the `ESP32-CAM` board [here](https://github.com/ivmarkov/rust-esp32-cam)
* I strongly suggest you use it as a base
* Change your Wifi ssid/pass in `src/lib.rs` first
* It does not really "map" any functionality specific to the camera on that board. For that, you need to manually crate unsafe `extern "C"` bindings in the `src/lib.rs` file. Given how few the methods of the CAM driver are, this is the easiest way to start. Don't think about "bindgen" as of yet.

> At least any suggestion or good starting point with rust esp32 for `bluetooth` ( after the idf menu config enabled ) & `espNow` communication to make them singing ?.

This is a whole project by itself, which is not easy, because there are **NO** standard, safe Rust APIs for Bluetooth, to which to map  the ESP32 C-level drivers. And my limited free time is completely occupied with other Rust embedded issues right now, so here you are on your own...

Dito for `espNow`.

Vandell63, 2021-09-24:
Thank for all. 
I'll give a shot for the esp32-cam starter project.
Let me know if can help you for any subjects.
I closed it for now.

closed 2021-09-24 by Vandell63