esp-rs/embuild #25: ldproxy doesn't build under windows

grbd, 2021-11-27:
Hi,
I recently tried installing ldproxy for use with the esp32 wifi demo
over at https://github.com/ivmarkov/rust-esp32-std-demo

However during the install under windows I ended up with around 6 of the below errors scattered about in different files
```
cargo install ldproxy

...

error[E0532]: expected tuple struct or tuple variant, found function `Ok`
   --> C:\Users\ric\.cargo\registry\src\github.com-1ecc6299db9ec823\embuild-0.25.3\src\utils.rs:199:13
    |
199 |             Ok(result) => {
    |             ^^ not a tuple struct or tuple variant
    |
   ::: C:\Users\ric\.cargo\registry\src\github.com-1ecc6299db9ec823\embuild-0.25.3\src\python.rs:12:23
    |
12  |     let version_str = cmd_output!(PYTHON, "--version")
    |                       -------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `cmd_output` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing one of these items instead
    |
1   | use core::result::Result::Ok;
    |
1   | use serde::__private::Ok;
    |
1   | use std::result::Result::Ok;
    |
```

So I tried adding in the following to the files where it was complaining
```
use std::result::Result::Ok;
```

That got me as far as a built ldproxy.exe
However when trying to build the rust-esp32-std-demo
The end result was
```
 = note: Running ldproxy
          thread 'main' panicked at 'Cannot locate argument '--ldproxy-linker <linker>'', ldproxy\src\main.rs:44:13
          note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

So I'm not sure if something has changed recently or if adding in that line was the wrong approach
Tried this on both the stable and nightly windows toolchains (stable-x86_64-pc-windows-msvc)

grbd, 2021-11-27:
Ok woop I just managed to get the example to build under windows with the above changes so I'll put a pull request in for it
Just needed to patch the depends in the example to use my local version

Strangely enough I also had to add the following to the top of src/main.rs within rust-esp32-std-demo to get it to build.

```
use std::result::Result::Ok;
```

ivmarkov, 2021-11-27:
See comment in https://github.com/ivmarkov/embuild/pull/26

grbd, 2021-11-27:
yep closing this down as this seems to be a compiler bug

grbd, 2021-11-27:
I've re-opened this for now since it seems to be an issue also with the stable toolchain.
I think it's a combination of an issue with the use of preludes inside a macro

N3xed, 2021-11-27:
Okay, I've dug into this and it seems that this comes from a regression of `anyhow` between version `1.0.43` and `1.0.49`.

Specifically, they added a function `Ok` with [this](https://github.com/dtolnay/anyhow/commit/2d4bb643b8a6f6095f93540cdc2bd7afd1cd4a12) commit, and because we import everything with `use anyhow::*` the `Ok` function shadows the tuple variant `Ok`.

closed 2021-11-28 by N3xed