esp-rs/embuild #30: "Ok(result) => {" not a tuple struct or tuple variant

JasonKleban, 2021-12-19:
[In this repo](https://github.com/JasonKleban/rust-esp32), having followed the steps in my readme.md, I'm getting these errors about the Result type in embuild source.  Any help please?

```
error: could not compile `embuild` due to 7 previous errors
warning: build failed, waiting for other jobs to finish...
error: build failed
error[E0532]: expected tuple struct or tuple variant, found function `Ok`
   --> C:\Users\Jason\.cargo\registry\src\github.com-1ecc6299db9ec823\embuild-0.24.5\src\utils.rs:123:13
    |
123 |             Ok(result) => {
    |             ^^ not a tuple struct or tuple variant
    |
   ::: C:\Users\Jason\.cargo\registry\src\github.com-1ecc6299db9ec823\embuild-0.24.5\src\bindgen.rs:164:5
    |
164 |     cmd!("rustup", "run", "stable", "rustfmt", output_file)?;
    |     ------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `cmd` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing one of these items instead
    |
1   | use serde::__private::Ok;
    |
1   | use std::result::Result::Ok;
    |
1   | use core::result::Result::Ok;
```

I started reporting with espflash because I figured it was something there, but that was denied:

https://github.com/esp-rs/espflash/issues/109#issuecomment-997466755

ivmarkov, 2021-12-20:
Please upgrade all dependencies to their latest versions. Not just embuild, but all esp-idf-* crates too.

I think you are hitting the glob-imported anyhow in embuild that we fixed a month ago.

closed 2021-12-20 by JasonKleban

ukd1, 2022-08-24:
@skerkour would you take a pull for the updates to this, as it seems this error happens when running through [your tutorial](https://kerkour.com/rust-on-esp32)?