esp-rs/embuild #42: [cargo install cargo-pio] error: failed to run custom build command for `ring v0.16.20`

Diegorro98, 2022-02-11:
Hi!

I've been trying to install cargo-pio, but I always have an esrror on build stage with ring:

```
error: failed to run custom build command for `ring v0.16.20`

Caused by:
  process didn't exit successfully: `C:\Users\1\AppData\Local\Temp\cargo-installzHzvXk\release\build\ring-f2ed3a1cac281d3d\build-script-build` (exit code: 101)
  --- stdout
  OPT_LEVEL = Some("3")
  TARGET = Some("x86_64-pc-windows-gnu")
  HOST = Some("x86_64-pc-windows-gnu")
  CC_x86_64-pc-windows-gnu = None
  CC_x86_64_pc_windows_gnu = None
  HOST_CC = None
  CC = None
  CFLAGS_x86_64-pc-windows-gnu = None
  CFLAGS_x86_64_pc_windows_gnu = None
  HOST_CFLAGS = None
  CFLAGS = None
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("false")
  CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")

  --- stderr
  running "gcc.exe" "-O3" "-ffunction-sections" "-fdata-sections" "-m64" "-I" "include" "-Wall" "-Wextra" "-std=c1x" "-Wbad-function-cast" "-Wnested-externs" "-Wstrict-prototypes" "-pedantic" "-pedantic-errors" "-Wall" "-Wextra" "-Wcast-align" "-Wcast-qual" "-Wconversion" "-Wenum-compare" "-Wfloat-equal" "-Wformat=2" "-Winline" "-Winvalid-pch" "-Wmissing-field-initializers" "-Wmissing-include-dirs" "-Wredundant-decls" "-Wshadow" "-Wsign-compare" "-Wsign-conversion" "-Wundef" "-Wuninitialized" "-Wwrite-strings" "-fno-strict-aliasing" "-fvisibility=hidden" "-g3" "-DNDEBUG" "-c" "-oC:\\Users\\1\\AppData\\Local\\Temp\\cargo-installzHzvXk\\release\\build\\ring-85838aa452798cfa\\out\\aes_nohw.o" "crypto/fipsmodule/aes/aes_nohw.c"
  crypto/fipsmodule/aes/aes_nohw.c:1:0: sorry, unimplemented: 64-bit mode not compiled in
   /* Copyright (c) 2019, Google Inc.

  thread 'main' panicked at 'execution failed', C:\Users\1\.cargo\registry\src\github.com-1ecc6299db9ec823\ring-0.16.20\build.rs:656:9
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: failed to compile `cargo-pio v0.25.2`, intermediate artifacts can be found at `C:\Users\1\AppData\Local\Temp\cargo-installzHzvXk`

Caused by:
  build failed
```

ivmarkov, 2022-02-11:
How did you install your rust compiler? Are you using rustup? If not, please switch to rustup. If yes, what rust compiler did you download? Can you please try with the non-gnu one?

Diegorro98, 2022-02-11:
Yes I used rustup

To make sure I used correctly I'm going to explplain my installation process:
Downloaded .exe from https://rustup.rs/ and then executed
Selected `Proceed with installation (default)` option
I get the following output:
```
info: profile set to 'default'
info: default host triple is x86_64-pc-windows-msvc
warning: Updating existing toolchain, profile choice will be ignored
info: syncing channel updates for 'stable-x86_64-pc-windows-gnu'
info: default toolchain set to 'stable-x86_64-pc-windows-gnu'

       stable-x86_64-pc-windows-gnu unchanged - rustc 1.58.1 (db9d1b20b 2022-01-20)

Rust is installed now. Great!                                                                                                                                                                                                                   To get started you may need to restart your current shell. 
This would reload its PATH environment variable to include
Cargo's bin directory (%USERPROFILE%\.cargo\bin).  
```

Is "stable-x86_64-pc-windows-gnu" the thing you are sugesting me to try with the non-gnu one?

ivmarkov, 2022-02-11:
info: default host triple is x86_64-pc-windows-msvc
**warning: Updating existing toolchain, profile choice will be ignored**
info: syncing channel updates for 'stable-x86_64-pc-windows-gnu'
info: default toolchain set to 'stable-x86_64-pc-windows-gnu

The text in bold is your problem. You are not starting from clean, and somehow as a result it re-uses your existing toolchain for Windows, which is the GNU one, instead of using the MSVC one.

I suggest you delete or temporarily rename your `.rustup` and `.cargo` directories and then try again.

Diegorro98, 2022-02-12:
Hey, it worked!
Thanks for your help :)

closed 2022-02-12 by Diegorro98