esp-rs/embuild #44: Add docs and a user friendly README

svenstaro, 2022-04-24:
Currently as a user coming into this project, I have 0 idea what embuild is meant to do for me as a whole. Could you add an overall README?

ivmarkov, 2022-04-24:
Sure. When I get to it. PRs greatly appreciated though!

svenstaro, 2022-04-24:
But I don't actually know the scope nor goal hence this issue. :D

ivmarkov, 2022-04-24:
Then this will stay open until I or someone else will fi d the time to invest into it. The point is, we really also need the help of the community in some of these things. Perhaps *this* particular issue is not the best example (docu of unknown code etc.) but there are a lot of other places where folks can help.

N3xed, 2022-07-20:
Fixed by 653f70393c096625c8a5b953c307b03d87f5e08c and 14c2bb64fa3dc083824c559d40dd054f306a5376.
There is always room for improvement, but this should suffice for now.

closed 2022-07-20 by N3xed