esp-rs/embuild #63: Issue building esp-idf

mfiumara, 2022-09-13:
Hi,

First of all thanks for the effort in creating embuild! I have been trying to use it to compile the example template application at https://github.com/esp-rs/esp-idf-template but I'm having some issues. When compiling I get the following error:

```
 In file included from C:/Users/matti/Documents/repos/wireprobe-rs/.embuild/espressif/esp-idf/release-v4.4/components/asio/asio/asio/include/asio/error_code.hpp:25,
                   from C:/Users/matti/Documents/repos/wireprobe-rs/.embuild/espressif/esp-idf/release-v4.4/components/asio/asio/asio/include/asio/error.hpp:19,
                   from C:/Users/matti/Documents/repos/wireprobe-rs/.embuild/espressif/esp-idf/release-v4.4/components/asio/asio/asio/include/asio/impl/error.ipp:20,
                   from C:/Users/matti/Documents/repos/wireprobe-rs/.embuild/espressif/esp-idf/release-v4.4/components/asio/asio/asio/include/asio/impl/src.hpp:22,
                   from C:/Users/matti/Documents/repos/wireprobe-rs/.embuild/espressif/esp-idf/release-v4.4/components/asio/asio/asio/src/asio.cpp:11:
  c:\users\matti\documents\repos\wireprobe-rs\.embuild\espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\riscv32-esp-elf\include\c++\8.4.0\system_error:39:10: fatal error: bits/error_constants.h: No such file or directory
   #include <bits/error_constants.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~
  compilation terminated.
  [698/1002] Linking C static library esp-idf\driver\libdriver.a
  [699/1002] Building C object esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cbortojson.c.obj
  [700/1002] Building C object esp-idf/cbor/CMakeFiles/__idf_cbor.dir/tinycbor/src/cborparser.c.obj
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Branch(
                  "release/v4.4",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: EspIdfRemote { repo_url: None, git_ref: Branch("release/v4.4") }
  fatal: No tags can describe 'de1e58118d89965f255edb587aae8dd29e8b510c'.
  Try --always, or create some tags.
  Using esp-idf v4.4.2 at 'C:\Users\matti\Documents\repos\wireprobe-rs\.embuild\espressif\esp-idf\release-v4.4'
  fatal: No tags can describe 'de1e58118d89965f255edb587aae8dd29e8b510c'.
  Try --always, or create some tags.
  thread 'main' panicked at '
  command did not execute successfully, got: exit code: 1

  build script failed, must exit now', C:\Users\matti\.cargo\registry\src\github.com-1ecc6299db9ec823\cmake-0.1.48\src\lib.rs:975:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

I'm not sure where to start looking, I have installed clang and installed rust using rustup,:
```
PS C:\Users\matti\Documents\repos\wireprobe-rs> rustc --version
rustc 1.65.0-nightly (fa6ee9375 2022-09-12)
PS C:\Users\matti\Documents\repos\wireprobe-rs> clang --version
clang version 15.0.0
Target: x86_64-pc-windows-msvc
Thread model: posix
InstalledDir: C:\Program Files\LLVM\bin
```

Any ideas where to start looking?

ivmarkov, 2022-09-14:
Shooting in the dark but can you sborten the name of your project, put it in drive C root and try again?

mfiumara, 2022-09-14:
Oh wow that actually worked. Is this a Windows feature? :P

I'll close this ticket thanks for the response!

closed 2022-09-14 by mfiumara

ivmarkov, 2022-09-15:
> Oh wow that actually worked. Is this a Windows feature? :P
> 
> I'll close this ticket thanks for the response!

Windows has a limit on path length and when file paths cross this limit weird issues like that start popping up. Not related to Rust dev.