esp-rs/embuild #77: is pio working?

camelator, 2023-07-08:
I just created a new project following the instructions here:
https://crates.io/crates/cargo-pio

my plateform.ini file is the following one, and the compilation fails with this error message:
/home/christian/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/debug/libFrameworkArduino.a(main.cpp.o): in function `main':
/home/christian/.platformio/packages/framework-arduinoststm32/cores/arduino/main.cpp:52: undefined reference to `setup'
/home/christian/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /home/christian/.platformio/packages/framework-arduinoststm32/cores/arduino/main.cpp:58: undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pio/build/debug/firmware.elf] Error 1

but the function is found in file main.cpp and lib.rs:

//
// Arduino main stub file. Calls into Rust.
//

extern "C" void arduino_setup();
extern "C" void arduino_loop();

void setup() {
    arduino_setup();
}

void loop() {
    arduino_loop();
}

lib.rs:
// Remove if STD is supported for your platform and you plan to use it
#![no_std]

// Remove if STD is supported for your platform and you plan to use it
#[panic_handler]
fn panic(_info: &core::panic::PanicInfo) -> ! {
    loop {}
}

//
// Entry points
//

#[no_mangle]
extern "C" fn arduino_setup() {
}

#[no_mangle]
extern "C" fn arduino_loop() {
}

; PlatformIO Project Configuration File
;
; Please visit documentation for options and examples
; https://docs.platformio.org/page/projectconf.html
[platformio]
default_envs = debug

[env]
extra_scripts = pre:platformio.git.py, pre:platformio.patch.py, platformio.cargo.py
board = genericSTM32F410C8
platform = ststm32
framework = arduino
rust_lib = test_rust
rust_target = thumbv7em-none-eabihf

[env:debug]
build_type = debug

[env:release]
build_type = release

camelator, 2023-07-08:
ok... Found the issue:
main.cpp is not complete:
#include <Arduino.h> is required at the beginning of the file

ivmarkov, 2023-07-09:
I've added the `#include <Arduino.h>` statement. The reason why it is necessary is because the Arduino framework for some MCUs have `setup`/`loop` defined as "C" functions, while for other frameworks, these are defined as "C++" functions. Hence the linkage errors.

closed 2023-07-09 by ivmarkov