esp-rs/embuild #81: `embuild` shouldn't require `git` or `.git` to exist in order to build the project

knarkzel, 2023-10-11:
When trying following in NixOS, it fails because this crate expects `.git` to exist for `$ESP_IDF`: https://github.com/knarkzel/hello-esp32c3

```
hello-esp32c3 via nix-shell-env ❯ cargo build
   Compiling esp-idf-sys v0.33.2
The following warnings were emitted during compilation:

warning: Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2, esp32h2, esp32c5]; explicitly specify an MCU to resolve this ambiguity

error: failed to run custom build command for `esp-idf-sys v0.33.2`

Caused by:
  process didn't exit successfully: `/home/odd/source/hello-esp32c3/target/debug/build/esp-idf-sys-c7e3a3706a45d51d/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:warning=Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2, esp32h2, esp32c5]; explicitly specify an MCU to resolve this ambiguity

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/nix/store/bmckqqb26qy16agaql75m5w1bwi4kp51-esp-idf-v5.1",
          ),
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  fatal: not a git repository (or any of the parent directories): .git
  fatal: not a git repository (or any of the parent directories): .git
  Error: '/nix/store/bmckqqb26qy16agaql75m5w1bwi4kp51-esp-idf-v5.1' is not a git respository

  Caused by:
      0: command 'cd "/nix/store/bmckqqb26qy16agaql75m5w1bwi4kp51-esp-idf-v5.1" && LANGUAGE="" LC_ALL="C.UTF-8" "git" "rev-parse" "--show-toplevel"' exited with non-zero status code 128
      1: fatal: not a git repository (or any of the parent directories): .git
```

ivmarkov, 2023-10-12:
[Here](https://github.com/esp-rs/embuild/pull/80)'s a pull request that was already reviewed and requires just a little love from somebody to get merged. (As in `cargo fmt`.) I guess the contributor lost interest as she is not replying.

If you would like to own it, we can have this issue fixed.

ivmarkov, 2023-10-12:
And the [corresponding PR to `esp-idf-sys`](https://github.com/esp-rs/esp-idf-sys/pull/241) that I can merge once the above PR is merged.

knarkzel, 2023-10-13:
Awesome, I'll give it a shot

knarkzel, 2023-10-13:
:laughing: 

[image]

ivmarkov, 2023-10-13:
@knarkzel Sorry - I've deleted your post. You know it could be anything as to why the previous contributor has not reacted.

Let's concentrate on owning and fixing the PR instead. :)

knarkzel, 2023-10-13:
Ait, so I tried out these patches:
```
[patch.crates-io]
esp-idf-sys = { git = "https://github.com/haileys/esp-idf-sys.git", rev = "211af868958ed5f5ceb2b7945a56bb131fd6ec11" }
embuild = { git = "https://github.com/haileys/embuild.git", rev = "5292871f5b3a0a78f398f80e8c159f47afb1df87" }
```
Output of `cargo tree`: https://pastebin.com/raw/f1rpDEwp

Still get the same error, am I missing something obvious?

```
  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/nix/store/bmckqqb26qy16agaql75m5w1bwi4kp51-esp-idf-v5.1",
          ),
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  fatal: not a git repository (or any of the parent directories): .git
  Using activated esp-idf v5.1.0 environment at '/nix/store/bmckqqb26qy16agaql75m5w1bwi4kp51-esp-idf-v5.1'
  thread 'main' panicked at /home/odd/.cargo/git/checkouts/esp-idf-sys-47955e843e87bb38/211af86/build/native/cargo_driver.rs:261:13:
  tree must always be Git variant if is_managed_espidf is true
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
Is it this `is_managed_espidf` that must be set?

knarkzel, 2023-10-13:
https://github.com/haileys/esp-idf-sys/commit/c4b8098b5201a8d7b3ac508edbf1086778f7cc56#diff-34a4c2047b858b330b2ca645dea0213b084e260f61c03c5288b65e2a395d3e9cR188-R189

knarkzel, 2023-10-13:
[image]

Need to run this custom path instead, what's the env variable for that?

ivmarkov, 2023-10-13:
`IDF_PATH`

knarkzel, 2023-10-13:
```
hello-esp32c3 via nix-shell-env ❯ echo $IDF_PATH
/nix/store/bmckqqb26qy16agaql75m5w1bwi4kp51-esp-idf-v5.1
```

knarkzel, 2023-10-13:
It still thinks it's managed despite having this.

ivmarkov, 2023-10-13:
Well you might have to dig deeper. As in setting `export RUST_BACKTRACE=1` and then re-running the build. This would give you an idea where exactly in the `esp-idf-sys` Rust build code it break, and you can look what is going on there.

knarkzel, 2023-10-13:
Hmm lets see...
```
  tree must always be Git variant if is_managed_espidf is true
  stack backtrace:
     0: rust_begin_unwind
               at /rustc/97c81e1b537088f1881c8894ee8579812ed9b6d1/library/std/sr     1: core::panicking::panic_fmt
               at /rustc/97c81e1b537088f1881c8894ee8579812ed9b6d1/library/core/s     2: build_script_build::native::cargo_driver::build
               at ./build/native/cargo_driver.rs:261:13
     3: build_script_build::native::build
               at ./build/native.rs:17:9
     4: build_script_build::main
               at ./build/build.rs:52:24
     5: core::ops::function::FnOnce::call_once
               at /rustc/97c81e1b537088f1881c8894ee8579812ed9b6d1/library/core/s
```

knarkzel, 2023-10-13:
I must be failing to patch something because the PR looks good

knarkzel, 2023-10-13:
Trying this:

[image]

Gives:

```
  fatal: not a git repository (or any of the parent directories): .git
  Using activated esp-idf v5.1.0 environment at '/nix/store/bmckqqb26qy16agaql75m5w1bwi4kp51-esp-idf-v5.1'
  fatal: not a git repository (or any of the parent directories): .git
  /nix/store/bmckqqb26qy16agaql75m5w1bwi4kp51-esp-idf-v5.1/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
    import pkg_resources
  fatal: not a git repository (or any of the parent directories): .git
  fatal: not a git repository (or any of the parent directories): .git
  Built components: riscv, esp_ringbuf, efuse, driver, esp_pm, mbedtls, esp_app_format, bootloader_support, bootloader, esptool_py, partition_table, esp_partition, app_update, esp_mm, spi_flash, pthread, esp_system, esp_rom, hal, log, heap, soc, esp_hw_support, freertos, newlib, cxx, esp_common, esp_timer, app_trace, esp_event, nvs_flash, esp_phy, vfs, lwip, esp_netif_stack, esp_netif, wpa_supplicant, esp_coex, esp_wifi, bt, unity, cmock, console, http_parser, esp-tls, esp_adc, esp_eth, esp_gdbstub, esp_hid, tcp_transport, esp_http_client, esp_http_server, esp_https_ota, esp_https_server, esp_psram, esp_lcd, protobuf-c, protocomm, esp_local_ctrl, espcoredump, wear_levelling, sdmmc, fatfs, idf_test, ieee802154, json, mqtt, openthread, spiffs, ulp, usb, wifi_provisioning, main
  thread 'main' panicked at /home/odd/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.63.0/./lib.rs:2338:31:
  Unable to find libclang: "couldn't find any valid shared libraries matching: ['libclang.so', 'libclang-*.so', 'libclang.so.*', 'libclang-*.so.*'], set the `LIBCLANG_PATH` environment variable to a path where one of these files can be found (invalid: [])"
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
Adding `clang`, `libclang` with `LIBCLANG_PATH = "${pkgs.libclang.lib}/lib"` then gives:
```
  Caused by:
      clang diagnosed error: /nix/store/y8wfrgk7br5rfz4221lfb9v8w3n0cnyd-glibc-2.37-8-dev/include/limits.h:145:5: error: function-like macro '__GLIBC_USE' is not defined
      /nix/store/y8wfrgk7br5rfz4221lfb9v8w3n0cnyd-glibc-2.37-8-dev/include/limits.h:184:5: error: function-like macro '__GLIBC_USE' is not defined
```

knarkzel, 2023-10-13:
```
   Compiling esp-idf-sys v0.33.2 (https://github.com/knarkzel/esp-idf-sys.git?branch=hailey#7c0fac0e)
The following warnings were emitted during compilation:

warning: Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2]; explicitly specify an MCU to resolve this ambiguity

error: failed to run custom build command for `esp-idf-sys v0.33.2 (https://github.com/knarkzel/esp-idf-sys.git?branch=hailey#7c0fac0e)`

Caused by:
  process didn't exit successfully: `/home/odd/source/nix/hello-esp32c3/target/debug/build/esp-idf-sys-26371692b110b770/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  cargo:warning=Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2]; explicitly specify an MCU to resolve this ambiguity
  cargo:rerun-if-changed=/home/odd/source/nix/hello-esp32c3/sdkconfig.defaults
  CMAKE_PREFIX_PATH_riscv32imc_zicsr_zifencei-esp-espidf = None
  CMAKE_PREFIX_PATH_riscv32imc_zicsr_zifencei_esp_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = Some("/nix/store/ay0p9mbw1w3zkvwzx3c94xq7x8jrn9wq-patchelf-0.15.0:/nix/store/yllcj971mwa0rvgwvd876a52391m8s9y-update-autotools-gnu-config-scripts-hook:/nix/store/18bs92p6yf6w2wwxhbplgx02y6anq092-gcc-wrapper-12.3.0:/nix/store/mc6q3cdz5s0p1aj4y586bglsfsnsf2k8-binutils-wrapper-2.40:/nix/store/bmckqqb26qy16agaql75m5w1bwi4kp51-esp-idf-v5.1:/nix/store/x3yahprgjd3arxxgbq4d3bxgl3hbghcg-python3-3.10.12-env:/nix/store/7smkn8cidx17la9ny5vfvghrwfxwbrny-git-2.42.0:/nix/store/g7hr76vc0bi7lb0qf9izvrblf0vh94m2-wget-1.21.4:/nix/store/5rdxplg3cczslq4banvk3449ry5k9hx7-gnumake-4.4.1:/nix/store/vlvzar852vc5i73hsr2gljh75lyjrpcp-flex-2.6.4:/nix/store/0v0b4pcb7fcjz693kqqhaci14l4zhrpr-gnum4-1.4.19:/nix/store/9jxcg001xr5h60r34zwdwmwkz7av1grb-bison-3.8.2:/nix/store/s15h6hdgzf60vqh2rp77vnbg58p1lv8i-gperf-3.1:/nix/store/5daca24rn22c65ff25lc6z0g0imfphvr-pkg-config-wrapper-0.29.2:/nix/store/yi235g10sp8jx939zpfli0s74154ph3v-cmake-3.26.4:/nix/store/ilgv1hna6y3cdsbi3vq87gvp7jpy6sqp-ninja-1.11.1:/nix/store/w0mmh06q0daxh4h544y1vdwqxbbgfmc1-ncurses-abi5-compat-6.4-dev:/nix/store/19x177m8wcf7c3q3mbc6vj08pbpy6lpj-ncurses-abi5-compat-6.4:/nix/store/7fnpn3lcf7wrwck9hrdkvr2913zykhij-dfu-util-0.11:/nix/store/ldbqi0jlrk23p4b5a3cl6ma34r7gw33v-riscv32-esp-elf-esp-idf-v5.1:/nix/store/bmmgfib1s1yl2fgkcvxwqhwzd4w6bwik-openocd-esp32-esp-idf-v5.1:/nix/store/9mnn1z71jzb39c3n8gk7fcq7lxfvm1da-riscv32-esp-elf-gdb-esp-idf-v5.1:/nix/store/sc6xfalvvq13vb4s9pzn6vcp59sjjb91-clang-wrapper-11.1.0:/nix/store/jcjxfkwypgx5w559vqc1z8pnkvr530g3-binutils-wrapper-2.40:/nix/store/y8wfrgk7br5rfz4221lfb9v8w3n0cnyd-glibc-2.37-8-dev:/nix/store/f6in5kb2y5v06zinz1a6xy6cyg67q026-glibc-2.37-8-bin:/nix/store/ld03l52xq2ssn4x0g5asypsxqls40497-glibc-2.37-8:/nix/store/c46mqx6v66daf2hvm22bczc6vnl02wq8-clang-11.1.0-dev:/nix/store/7y67h7q26mh2zwhilcpq82553s5bda21-clang-11.1.0-lib:/nix/store/71xzjk8qi1cjkwr8lv9rvk7s0zjiipzm-clang-11.1.0:/nix/store/63a68c8lz0ybq7mzyphx74g91ihqn90h-rust-default-1.75.0-nightly-2023-10-08:/nix/store/swprckq7jbx336bfcjvp7h9jdhid43d7-compiler-rt-libc-11.1.0-dev:/nix/store/ncm999fc8hxgazhgr0gswy2pxd0vs9sb-compiler-rt-libc-11.1.0")
  CMAKE_riscv32imc_zicsr_zifencei-esp-espidf = None
  CMAKE_riscv32imc_zicsr_zifencei_esp_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: cd "/home/odd/source/nix/hello-esp32c3/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-cc9e614875941005/out/build" && CMAKE_PREFIX_PATH="/nix/store/ay0p9mbw1w3zkvwzx3c94xq7x8jrn9wq-patchelf-0.15.0:/nix/store/yllcj971mwa0rvgwvd876a52391m8s9y-update-autotools-gnu-config-scripts-hook:/nix/store/18bs92p6yf6w2wwxhbplgx02y6anq092-gcc-wrapper-12.3.0:/nix/store/mc6q3cdz5s0p1aj4y586bglsfsnsf2k8-binutils-wrapper-2.40:/nix/store/bmckqqb26qy16agaql75m5w1bwi4kp51-esp-idf-v5.1:/nix/store/x3yahprgjd3arxxgbq4d3bxgl3hbghcg-python3-3.10.12-env:/nix/store/7smkn8cidx17la9ny5vfvghrwfxwbrny-git-2.42.0:/nix/store/g7hr76vc0bi7lb0qf9izvrblf0vh94m2-wget-1.21.4:/nix/store/5rdxplg3cczslq4banvk3449ry5k9hx7-gnumake-4.4.1:/nix/store/vlvzar852vc5i73hsr2gljh75lyjrpcp-flex-2.6.4:/nix/store/0v0b4pcb7fcjz693kqqhaci14l4zhrpr-gnum4-1.4.19:/nix/store/9jxcg001xr5h60r34zwdwmwkz7av1grb-bison-3.8.2:/nix/store/s15h6hdgzf60vqh2rp77vnbg58p1lv8i-gperf-3.1:/nix/store/5daca24rn22c65ff25lc6z0g0imfphvr-pkg-config-wrapper-0.29.2:/nix/store/yi235g10sp8jx939zpfli0s74154ph3v-cmake-3.26.4:/nix/store/ilgv1hna6y3cdsbi3vq87gvp7jpy6sqp-ninja-1.11.1:/nix/store/w0mmh06q0daxh4h544y1vdwqxbbgfmc1-ncurses-abi5-compat-6.4-dev:/nix/store/19x177m8wcf7c3q3mbc6vj08pbpy6lpj-ncurses-abi5-compat-6.4:/nix/store/7fnpn3lcf7wrwck9hrdkvr2913zykhij-dfu-util-0.11:/nix/store/ldbqi0jlrk23p4b5a3cl6ma34r7gw33v-riscv32-esp-elf-esp-idf-v5.1:/nix/store/bmmgfib1s1yl2fgkcvxwqhwzd4w6bwik-openocd-esp32-esp-idf-v5.1:/nix/store/9mnn1z71jzb39c3n8gk7fcq7lxfvm1da-riscv32-esp-elf-gdb-esp-id [... cut ...]
  Error: failed to generate bindings in file '/home/odd/source/nix/hello-esp32c3/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-cc9e614875941005/out/bindings.rs'

  Caused by:
      clang diagnosed error: /nix/store/y8wfrgk7br5rfz4221lfb9v8w3n0cnyd-glibc-2.37-8-dev/include/limits.h:145:5: error: function-like macro '__GLIBC_USE' is not defined
      /nix/store/y8wfrgk7br5rfz4221lfb9v8w3n0cnyd-glibc-2.37-8-dev/include/limits.h:184:5: error: function-like macro '__GLIBC_USE' is not defined
```

knarkzel, 2023-10-13:
It seems the pull request fixed one issue, but there's several other places that must be fixed

ivmarkov, 2023-10-13:
Maybe the relevant question here is... is `ESP IDF` **itself** supporting a build where "it" is not a GIT repository? Because if it doesn't, we should forget about this change altogether.

ivmarkov, 2023-10-13:
By the way... what is it with *nix that I cannot automatically clone a git repo and then just use it?

knarkzel, 2023-10-13:
It's possible, just not recommended. Everything in `/nix/store` is immutable. I tried quickly, but ran into some issue.

knarkzel, 2023-10-13:
I'll take another look later

knarkzel, 2023-10-13:
Got the same issue (with managed disabled) when using shell that lets `esp-idf-sys` build from scratch:

```
  /nix/store/y8wfrgk7br5rfz4221lfb9v8w3n0cnyd-glibc-2.37-8-dev/include/limits.h:145:5: error: function-like macro '__GLIBC_USE' is not defined
  /nix/store/y8wfrgk7br5rfz4221lfb9v8w3n0cnyd-glibc-2.37-8-dev/include/limits.h:184:5: error: function-like macro '__GLIBC_USE' is not defined
  /home/odd/source/nix/hello-esp32c3/.embuild/espressif/esp-idf/v4.4/components/freertos/port/riscv/include/freertos/portmacro.h:439:20: warning: section does not match previous declaration [-Wsection]
  /home/odd/source/nix/hello-esp32c3/.embuild/espressif/esp-idf/v4.4/components/freertos/port/riscv/include/freertos/portmacro.h:271:20: note: previous attribute is here
  /home/odd/source/nix/hello-esp32c3/.embuild/espressif/esp-idf/v4.4/components/esp_hw_support/include/soc/esp_spiram.h:1:2: warning: esp_spiram.h has been replaced by esp32/spiram.h, please include esp32/spiram.h instead [-W#warnings]
  clang diag: /home/odd/source/nix/hello-esp32c3/.embuild/espressif/esp-idf/v4.4/components/freertos/port/riscv/include/freertos/portmacro.h:439:20: warning: section does not match previous declaration [-Wsection]
  clang diag: /home/odd/source/nix/hello-esp32c3/.embuild/espressif/esp-idf/v4.4/components/esp_hw_support/include/soc/esp_spiram.h:1:2: warning: esp_spiram.h has been replaced by esp32/spiram.h, please include esp32/spiram.h instead [-W#warnings]
  Error: failed to generate bindings in file '/home/odd/source/nix/hello-esp32c3/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-b669bb536e7636c2/out/bindings.rs'

  Caused by:
      clang diagnosed error: /nix/store/y8wfrgk7br5rfz4221lfb9v8w3n0cnyd-glibc-2.37-8-dev/include/limits.h:145:5: error: function-like macro '__GLIBC_USE' is not defined
      /nix/store/y8wfrgk7br5rfz4221lfb9v8w3n0cnyd-glibc-2.37-8-dev/include/limits.h:184:5: error: function-like macro '__GLIBC_USE' is not defined
```

knarkzel, 2023-10-13:
This is the shell:

```nix
{
  inputs = {
    nixpkgs.url = "github:NixOS/nixpkgs/nixos-unstable";
    esp-dev = {
      url = "github:thiskappaisgrey/nixpkgs-esp-dev-rust";
      inputs.nixpkgs.follows = "nixpkgs";
    };
  };

  outputs = {
    self,
    nixpkgs,
    esp-dev,
  }: let
    pkgs = import nixpkgs {
      system = "x86_64-linux";
      overlays = [ esp-dev.overlay ];
    };
  in {
    devShells.x86_64-linux.default = pkgs.mkShell {
      buildInputs = with pkgs; [
        gcc-riscv32-esp32c3-elf-bin
        openocd-esp32-bin

        # Tools required to use ESP-IDF.
        git
        wget
        gnumake

        flex
        bison
        gperf
        pkg-config

        cmake
        ninja

        ncurses5

        python3
        python3Packages.pip
        python3Packages.virtualenv
      ];

      shellHook = ''
        export ESP_IDF_VERSION="v4.4"
        export LD_LIBRARY_PATH="${pkgs.lib.makeLibraryPath [ pkgs.libxml2 pkgs.zlib pkgs.stdenv.cc.cc.lib ]}"
        export LIBCLANG_PATH=${pkgs.llvmPackages.libclang.lib}/lib
      '';
    };
  };
}
```

knarkzel, 2023-10-13:
Might've forgotten to remove the patch, trying again.. :laughing:

ivmarkov, 2023-10-13:
> Maybe the relevant question here is... is `ESP IDF` **itself** supporting a build where "it" is not a GIT repository? Because if it doesn't, we should forget about this change altogether.

I think you should try to confirm/deny this first.

knarkzel, 2023-10-13:
Tried it where I let `embuild` download `esp-idf`, but I got to the same errors: https://github.com/esp-rs/esp-idf-sys/issues/184#issuecomment-1761444853

ivmarkov, 2023-10-13:
You need to git clone esp idf ONLY and then try to build something with it. Like a C program. If this does not work out of the box on *nix, whatever else we try to do in the Rust crates won't work either, as they automate exactly that.

denbeigh2000, 2024-11-25:
Hi @ivmarkov,

I'm interested in seeing this issue resolved, would you be open to reviewing another iteration of #80/#82 that's been rebased and formatted?

ivmarkov, 2024-11-25:
> Hi @ivmarkov,
> 
> I'm interested in seeing this issue resolved, would you be open to reviewing another iteration of #80/#82 that's been rebased and formatted?

Sure, but it is not just about rebasing and formatting. Someone has to confirm that it actually does work.

denbeigh2000, 2024-11-25:
> > Hi @ivmarkov,
> > I'm interested in seeing this issue resolved, would you be open to reviewing another iteration of #80/#82 that's been rebased and formatted?
> 
> Sure, but it is not just about rebasing and formatting. Someone has to confirm that it actually does work.

Okay, I'll attempt a validation build of `esp-idf-hal` with my fork this week.

denbeigh2000, 2024-11-26:
@ivmarkov I've confirmed _some_ success here.

I have cloned `esp-idf-sys`, and [updated it](https://github.com/esp-rs/esp-idf-sys/compare/master...denbeigh2000:esp-idf-sys:denbeigh/forked-embuild?expand=1) to use my own `embuild` fork. There are definitely some cleanups that could be done, but hopefully it'll serve as a proof-of-concept.

I tried to build an example in that repo: The crate itself compiles and locates the IDF_* paths from my environment - however, cmake appears to perform the same `rev-parse --show-toplevel` check when invoked, which then fails.

<details>
<summary>Collapsed cargo output
</summary>

```
$ MCU=esp32 cargo espflash flash --target xtensa-esp32-espidf --example std_basics

...

   Compiling cargo-platform v0.1.8
   Compiling envy v0.4.2
   Compiling cargo_metadata v0.18.1
   Compiling embuild v0.32.0 (https://github.com/denbeigh2000/embuild?rev=b8b924b7a5bdbe947d5fff252cd24f61bda00074#b8b924b7)
   Compiling proc_macro v0.0.0 (/nix/store/pbwyjfkxd4vw7wgmmzrlsnz1qzcijrds-rust-esp-v1.82.0.3/lib/rustlib/src/rust/library/proc_macro)
   Compiling const_format v0.2.33
   Compiling esp-idf-sys v0.35.0 (/Users/denbeigh/dev/esp-idf-sys)
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.35.0: (esp-idf-sys) Removed env vars that might affect the ESP IDF C build: `CC=clang;CXX=clang++`

error: failed to run custom build command for `esp-idf-sys v0.35.0 (/Users/denbeigh/dev/esp-idf-sys)`

Caused by:
  process didn't exit successfully: `/Users/denbeigh/dev/esp-idf-sys/target/debug/build/esp-idf-sys-b7d144c692d6a5ef/build-script-build` (exit status: 101)
  --- stdout
  cargo:warning=(esp-idf-sys) Removed env vars that might affect the ESP IDF C build: `CC=clang;CXX=clang++`
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  cargo:rerun-if-changed=/Users/denbeigh/dev/esp-idf-sys/.github/configs/sdkconfig.defaults
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: cd "/Users/denbeigh/dev/esp-idf-sys/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-4a843b26a86fe601/out/build" && CMAKE_PREFIX_PATH="" EXTRA_COMPONENT_DIRS="" IDF_COMPONENT_MANAGER="1" IDF_PATH="/nix/store/yfi52smvpn84vz4wxa0jh5na9yihd9hb-esp-idf-f420609c332fbd2d2f7f188c6579d046c9560e42" IDF_TARGET="esp32" PATH="/nix/store/pbwyjfkxd4vw7wgmmzrlsnz1qzcijrds-rust-esp-v1.82.0.3/bin:/nix/store/63f4im3v1vwr9hv3akppn1i5l8p0mpcc-python3-3.12.7-env/bin:/nix/store/w5rggkxjd7z2cjk4fia9za7gb485pydw-git-2.47.0/bin:/nix/store/3hfdiifrid2mf5v5w8va194bhgj08i57-wget-1.25.0/bin:/nix/store/0gqbh56qwj0cxmvl7dajic9mkbc26ilc-gnumake-4.4.1/bin:/nix/store/p7wqpzz9dprq33fbs70dlj1haac26871-flex-2.6.4/bin:/nix/store/2sk7w1y6pmz8b51290fj7jvglqb55skx-gnum4-1.4.19/bin:/nix/store/kykd5x1fc167yhf87pnsqyd17j6xaksk-bison-3.8.2/bin:/nix/store/154nqi968xpkrd6xk8vycyv9pgkpgawp-gperf-3.1/bin:/nix/store/c6v2ngygy55zamyz17z94brbz2vrsi1z-pkg-config-wrapper-0.29.2/bin:/nix/store/aw70f6alsbf7cgn5qwqbpgd1w208i3v9-cmake-3.30.5/bin:/nix/store/hj6jb9m7rp59bgj7r2jpbqw9yip8fapr-ninja-1.12.1/bin:/nix/store/33lq0qgjciw7h4x9zg5ws3mpk6d89wfc-ncurses-abi5-compat-6.4.20221231-dev/bin:/nix/store/h6bs7szrrdngpw5hy7w0wlvmdvi6yz7q-ncurses-abi5-compat-6.4.20221231/bin:/nix/store/m5fni2vq69nwb3w5ks1mzkw0ajcmsilj-dfu-util-0.11/bin:/nix/store/0ixmcknpjhjxbnys9bkv68rr44ljvf11-esp-clang-esp-idf-f420609c332fbd2d2f7f188c6579d046c9560e42/bin:/nix/store/jxi8kf9isy6jhzx2kq81csccifpydpwq-xtensa-esp-elf-esp-idf-f420609c332fbd2d2f7f188c6579d046c9560e42/bin:/nix/store/5qrqmkwcx48wg0bfcy77xlp7dzbmdk44-esp32ulp-elf-esp-idf-f420609c332fbd2d2f7f188c6579d046c9560e42/bin:/nix/store/0ycmpdkasf51xmjlp4l78grqx5dip9h6-openocd-esp32-esp-idf-f420609c332fbd2d2f7f188c6579d046c9560e42/bin:/nix/store/4zz75akpch21wypfv092gh6j3iz69bm6-xtensa-esp-elf-gdb-esp-idf-f420609c332fbd2d2f7f188c6579d046c9560e42/bin:/nix/store/n1zwfydcicy7az4xhmfzjsrqcdc6qmrx-openssl-3.3.2-bin/bin:/nix/store/v39c0h6xv6hvki2k1bsyp5n090q8y2bp-clang-wrapper-16.0.6/bin:/nix/store/44hqr4414m3jldixkm4cq47q5pnpy83d-clang-16.0.6/bin:/nix/store/5g6np23z9q7vwwp1s5pxkgn8f8wrmihh-coreutils-9.5/bin:/nix/store/jqy6x4li0s729q6q97qmzrqwjfi567bv-cctools-binutils-darwin-wrapper-1010.6/bin:/nix/store/c02f0lj5xqjaw1c6hg9a9sv1b024mi8q-cctools-binutils-darwin-1010.6/bin:/nix/store/3a99nvsn1jh7h3vklnd8py5jplls9q41-xcbuild-0.1.1-unstable-2019-11-20-xcrun/bin:/nix/store/iwbpxcjz1s816d80sqg0mxg5p8m0dqak-libiconv-107/bin:/nix/store/5g6np23z9q7vwwp1s5pxkgn8f8wrmihh-coreutils-9.5/bin:/nix/store/zxz2b4dlg9zz8dcrwa2my1bpyxpfcchj-findutils-4.10.0/bin:/nix/store/xnm74pr7sl0jvn2q8fcr4jy3vrdgyqrn-diffutils-3.10/bin:/nix/store/9m372k9wy0f9aj54zhvsr65w2p9ndwnj-gnused-4.9/bin:/nix/store/ay0r3wrwbv5rq3mg3vhla045yjr49gka-gnugrep-3.11/bin:/nix/store/5q48fvg6grf5zsk106cm0k0hbfc1dfp2-gawk-5.3.1/bin:/nix/store/41dh2jr0 [... cut ...]
  -- Check for working C compiler: /nix/store/jxi8kf9isy6jhzx2kq81csccifpydpwq-xtensa-esp-elf-esp-idf-f420609c332fbd2d2f7f188c6579d046c9560e42/bin/xtensa-esp32-elf-gcc
  -- Check for working C compiler: /nix/store/jxi8kf9isy6jhzx2kq81csccifpydpwq-xtensa-esp-elf-esp-idf-f420609c332fbd2d2f7f188c6579d046c9560e42/bin/xtensa-esp32-elf-gcc - broken
  -- Configuring incomplete, errors occurred!
```

</details>

What validation would you like to see? Is there a more isolated test I can do that would validate this work before similar changes are made in cmake tooling?

ivmarkov, 2024-11-26:
> @ivmarkov I've confirmed _some_ success here.
> 
> I have cloned `esp-idf-sys`, and [updated it](https://github.com/esp-rs/esp-idf-sys/compare/master...denbeigh2000:esp-idf-sys:denbeigh/forked-embuild?expand=1) to use my own `embuild` fork. There are definitely some cleanups that could be done, but hopefully it'll serve as a proof-of-concept.
> 
> I tried to build an example in that repo: The crate itself compiles and locates the IDF_* paths from my environment - however, cmake appears to perform the same `rev-parse --show-toplevel` check when invoked, which then fails.
> 
> Collapsed cargo output
> What validation would you like to see? Is there a more isolated test I can do that would validate this work before similar changes are made in cmake tooling?

Before merging this, I would like to see a successful build, even if that means forking the `cmake-rs` repo and doing the necessary changes there. :)

In fact, having a very clear picture what changes (if any? are we 100% sure we need to make changes to `cmake-rs` itself, and we cannot just drive it with configuration so that it does not do the GIT check?) is very important, because upstreaming changes to `cmake-rs` is a very long process.

We just waited ~ 1 month for a 3-lines' change to make it in there.

denbeigh2000, 2024-11-26:
> Before merging this, I would like to see a successful build, even if that means forking the `cmake-rs` repo and doing the necessary changes there. :)
>
> In fact, having a very clear picture what [...]

Okay, seems reasonable - I'll dig in more this week and see what I come up with.

Hoping my changes can be kept to `idf-sys` 🤞

denbeigh2000, 2024-11-29:
@ivmarkov: Okay! I've confirmed that with these changes, I can successfully build the `std_basics` example from `esp-idf-sys` in both scenarios:
- `IDF_PATH` etc use a path that is not a valid git repository
- `IDF_PATH` etc are unset

I've put my build information/output in the collapsed block below.

<details>
<summary>`IDF_PATH` -> Invalid git repository
</summary>

```
validation@d11c501155b9:~/esp-idf-sys$ cargo clean
     Removed 2989 files, 1.2GiB total
validation@d11c501155b9:~/esp-idf-sys$ env | grep IDF
IDF_PYTHON_ENV_PATH=/home/validation/.espressif/python_env/idf5.1_py3.12_env
IDF_PATH=/home/validation/esp/esp-idf
ESP_IDF_VERSION=5.1
IDF_DEACTIVATE_FILE_PATH=/tmp/tmpfzxdlsruidf_108858
IDF_TOOLS_EXPORT_CMD=/home/validation/esp/esp-idf/export.sh
IDF_TOOLS_INSTALL_CMD=/home/validation/esp/esp-idf/install.sh
validation@d11c501155b9:~/esp-idf-sys$ git log --pretty=oneline -1
b9df9f16d6dcc1efd086b5fb84cf672a77f12300 (HEAD -> denbeigh/forked-embuild, origin/denbeigh/forked-embuild) validation: use forked embuild
validation@d11c501155b9:~/esp-idf-sys$ pushd $IDF_PATH; git status; popd
~/esp/esp-idf ~/esp-idf-sys
fatal: not a git repository (or any of the parent directories): .git
~/esp-idf-sys
validation@d11c501155b9:~/esp-idf-sys$ MCU=esp32 cargo build --target xtensa-esp32-espidf --example std_basics
   Compiling proc-macro2 v1.0.92
   Compiling compiler_builtins v0.1.123
   Compiling unicode-ident v1.0.14
   Compiling core v0.0.0 (/home/validation/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling memchr v2.7.4
   Compiling cfg-if v1.0.0
   Compiling serde v1.0.215
   Compiling libc v0.2.166
   Compiling bitflags v2.6.0
   Compiling libc v0.2.158
   Compiling regex-syntax v0.8.5
   Compiling crossbeam-utils v0.8.20
   Compiling glob v0.3.1
   Compiling rustix v0.38.41
   Compiling cfg_aliases v0.2.1
   Compiling memchr v2.5.0
   Compiling rustversion v1.0.18
   Compiling nix v0.29.0
   Compiling prettyplease v0.2.25
   Compiling syn v1.0.109
   Compiling log v0.4.22
   Compiling linux-raw-sys v0.4.14
   Compiling serde_json v1.0.133
   Compiling minimal-lexical v0.2.1
   Compiling either v1.13.0
   Compiling same-file v1.0.6
   Compiling thiserror v1.0.69
   Compiling shlex v1.3.0
   Compiling walkdir v2.5.0
   Compiling libloading v0.8.5
   Compiling aho-corasick v1.1.3
   Compiling bstr v1.11.0
   Compiling nom v7.1.3
   Compiling clang-sys v1.8.1
   Compiling cvt v0.1.2
   Compiling semver v1.0.23
   Compiling quote v1.0.37
   Compiling crossbeam-epoch v0.9.18
   Compiling ryu v1.0.18
   Compiling syn v2.0.89
   Compiling crossbeam-deque v0.8.5
   Compiling once_cell v1.20.2
   Compiling heck v0.4.1
[... 214 lines cut ...]
    = note: see <https://doc.rust-lang.org/nightly/rustc/check-cfg/cargo-specifics.html> for more information about checking conditional configuration
    = note: `#[warn(unexpected_cfgs)]` on by default

warning: unexpected `cfg` condition name: `esp_idf_version`
  --> examples/std_basics.rs:31:15
   |
31 |     #[cfg(not(esp_idf_version = "4.3"))]
   |               ^^^^^^^^^^^^^^^^^^^^^^^
   |
   = help: consider using a Cargo feature instead
   = help: or consider adding in `Cargo.toml` the `check-cfg` lint config for the lint:
            [lints.rust]
            unexpected_cfgs = { level = "warn", check-cfg = ['cfg(esp_idf_version, values("4.3"))'] }
   = help: or consider adding `println!("cargo::rustc-check-cfg=cfg(esp_idf_version, values(\"4.3\"))");` to the top of the `build.rs`
   = note: see <https://doc.rust-lang.org/nightly/rustc/check-cfg/cargo-specifics.html> for more information about checking conditional configuration

warning: `esp-idf-sys` (example "std_basics") generated 2 warnings
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 1m 04s
```

</details>

If you're happy with this, I'll PR these changes.

After that, I can also PR the necessary [changes to `esp-idf-sys`](https://github.com/esp-rs/esp-idf-sys/compare/master...denbeigh2000:esp-idf-sys:denbeigh/forked-embuild?expand=1) (after tidying them up).

ivmarkov, 2024-11-29:
@denbeigh2000 OK, good!

Next step is to open PRs against the embuild and sys repos that contain your changes - once you feel ready with those.

Would be good to also explain shortly how you did workaround the failure in the CMake build. If I'm not mistaken, the original plan was to do changes to `cmake-rs`? Apparently, it is no longer mecessary, but also I can't see in your changes how exactly you silenced cmake not to complain that the ESP IDF dir is not a GIT repo?

denbeigh2000, 2024-11-29:
> @denbeigh2000 OK, good!
> 
> Next step is to open PRs against the embuild and sys repos that contain your changes - once you feel ready with those.

No problem, I'll get those up today.

> Would be good to also explain shortly how you did workaround the failure in the CMake build. If I'm not mistaken, the original plan was to do changes to `cmake-rs`? Apparently, it is no longer [...]

Sure - I think the _actual_ problem in my initial validation was user error, I misconfigured my environment variables while trying to package my toolchain. 😅 

---

The idea was never to tackle `cmake-rs`, but rather just the cmake configuration in `esp-idf` (if it was necessary).

This is the [specific error I was hitting](https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/tools/cmake/third_party/GetGitRevisionDescription.cmake#L46-L63) in my output: (`git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)`)

Going in further, I found this should only be [evaluated](https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/tools/cmake/idf.cmake#L38) if my [env vars are unset](https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/tools/cmake/idf.cmake#L1-L2). Once I realised that, I decided to just test the problem on its own.

Confusingly, IDF still attempts to call `git` several times - the output still pollutes stdout/stderr, but the compilation succeeds (there have been a few [recent changes to IDF](https://github.com/espressif/esp-idf/commit/23ba79d5c75d63c2a37d6481a4d4ac9d3b40d280) to make this non-fatal). This isn't seen in a successful `cargo build` because `cargo` hides stdout/stderr from successful builds.

---

Turns out compiler toolchains are hard, and solving just one problem at once made life a lot easier :smile:

denbeigh2000, 2024-11-29:
After doing some more thorough testing, I'd like one more day to consider these changes.

Namely, I want to fix the interaction between these functions. Currently, they both need to be aware of if `IDF_PATH` is externally provided, which feels wrong.
- https://github.com/denbeigh2000/esp-idf-sys/blob/c316174c0765494b7698cdfb926bebd630821ade/build/native/cargo_driver.rs#L182
- https://github.com/denbeigh2000/embuild/blob/be687ce7dc78fd4cba6f0defd513e0ebb32aac73/src/espidf.rs#L377

---

My changes to `esp-idf-sys` will need to be updated, and I will need to submit a change to `esp-idf`.

I found that I was not invoking `try_from_env` correctly [here](https://github.com/denbeigh2000/esp-idf-sys/blob/c316174c0765494b7698cdfb926bebd630821ade/build/native/cargo_driver.rs#L182), which meant `embuild` was always managing my `esp-idf` checkout.

When I fixed this, I that [this check in `openthread`](https://github.com/espressif/openthread/blob/005c5cefc22aaf0396e4327ee7f2e0ad32a7733b/CMakeLists.txt#L104-L108) also causes a build failure.

IMHO the nicest way to fix would be applying this change in [`esp-idf`](https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/components/openthread/CMakeLists.txt#L253-L263), which will avoid any changes to `openbuild`:

```diff
diff --git a/components/openthread/CMakeLists.txt b/components/openthread/CMakeLists.txt
index 455b346924..5b538c2d57 100644
--- a/components/openthread/CMakeLists.txt
+++ b/components/openthread/CMakeLists.txt
@@ -168,20 +168,26 @@ execute_process(
     COMMAND git rev-parse --short HEAD
     WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
     OUTPUT_VARIABLE IDF_VERSION_FOR_OPENTHREAD_PACKAGE OUTPUT_STRIP_TRAILING_WHITESPACE
 )

 execute_process(
     COMMAND git rev-parse --short HEAD
     WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}/openthread
     OUTPUT_VARIABLE OPENTHREAD_VERSION OUTPUT_STRIP_TRAILING_WHITESPACE
 )
+if (STATUS AND NOT STATUS EQUAL 0)
+    # If this is not a valid git checkout, eagerly set OT_PACKAGE_VERSION for openthread.
+    # If this is unset, openthread's compilation process will invoke git to set this variable,
+    # causing a build failure.
+    set(OT_PACKAGE_VERSION "development" CACHE STRING "OpenThread Package Version")
+endif()

 string(TIMESTAMP OT_BUILD_TIMESTAMP " %Y-%m-%d %H:%M:%S UTC" UTC)
 string(CONCAT OT_FULL_VERSION_STRING
        "openthread-esp32/"
        "${IDF_VERSION_FOR_OPENTHREAD_PACKAGE}-${OPENTHREAD_VERSION}\; "
        "${CONFIG_IDF_TARGET}\; ${OT_BUILD_TIMESTAMP}")

 idf_component_register(SRC_DIRS "${src_dirs}"
                        EXCLUDE_SRCS "${exclude_srcs}"
                        INCLUDE_DIRS "${public_include_dirs}"
```

That's the only change I've needed to make to ensure a successful compilation without `esp-idf` being a valid git repo.

ivmarkov, 2024-11-30:
Yes - sure - take your time. Non-GIT build support was in the queue for so long time, that even if it needs a couple of extra weeks, no big deal I guess.

I suggest you just mark one or both of your PRs as "Draft" until you feel they are ready for submission and review, and then you can un-"draft" them.

denbeigh2000, 2024-12-01:
I got pretty caught out by failures that were difficult to reproduce after the first successful build yesterday, so I reconsidered my approach to testing and made it a lot more repeatable.

I'm confident after [thorough testing](https://github.com/esp-rs/embuild/pull/95#issuecomment-2510228806) that this change is doing the right thing, and I've included annotated `stderr` of all my builds for proper validation.

---

Interestingly enough, I did run into the same issue with `openthread` performing `git` operations in its' own build script, but it didn't seem to cause a failure. I've lost the original `stderr` of the build I found that issue on, but I can successfully repeat the build configuration in my testing harness successfully, so 🤷.

PR #95 (2024-12-22): Support building with non-git repositories
This is a repackaging of the work originally done in #80 and #82.

This allows `embuild` to build non-git source repositories. This work will allow dependencies (such was `esp-idf`) to be provided by sources other than `embuild` that may not include the surrounding `git` repo, such as the Nix package manager.

Reasonable testing has been completed in the [comment below](https://github.com/esp-rs/embuild/pull/95#issuecomment-2510228806).

Closes #80
Closes #81
Closes #82
closed 2024-12-22 by ivmarkov