esp-rs/embuild #83: Won't build with esp-idf 5.2

clouds56, 2024-03-10:
Using template https://github.com/esp-rs/esp-template with `ESP_IDF_VERSION = "v5.2.1"`

```
  = note: [ldproxy] Running ldproxy
          Error: Linker /Users/clouds/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          /Users/clouds/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/clouds/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/lib/esp32s3/no-rtti/libstdc++.a(eh_alloc.o):(.literal.startup._GLOBAL__sub_I__ZN9__gnu_cxx9__freeresEv+0x10): undefined reference to `getenv'
          /Users/clouds/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/clouds/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/lib/esp32s3/no-rtti/libstdc++.a(eh_alloc.o): in function `_GLOBAL__sub_I__ZN9__gnu_cxx9__freeresEv':
          /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_alloc.cc:203:(.text.startup._GLOBAL__sub_I__ZN9__gnu_cxx9__freeresEv+0x1b): undefined reference to `getenv'
          collect2: error: ld returned 1 exit status
          
          
          Stack backtrace:
             0: std::backtrace_rs::backtrace::libunwind::trace
                       at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/../../backtrace/src/backtrace/libunwind.rs:104:5
             1: std::backtrace_rs::backtrace::trace_unsynchronized
                       at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
             2: std::backtrace::Backtrace::create
                       at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/backtrace.rs:331:13
             3: anyhow::error::<impl anyhow::Error>::msg
             4: ldproxy::main
             5: std::sys_common::backtrace::__rust_begin_short_backtrace
             6: std::rt::lang_start::{{closure}}
             7: core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
                       at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/core/src/ops/function.rs:284:13
             8: std::panicking::try::do_call
                       at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panicking.rs:552:40
             9: std::panicking::try
                       at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panicking.rs:516:19
            10: std::panic::catch_unwind
                       at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panic.rs:142:14
            11: std::rt::lang_start_internal::{{closure}}
                       at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/rt.rs:148:48
            12: std::panicking::try::do_call
                       at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panicking.rs:552:40
            13: std::panicking::try
                       at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panicking.rs:516:19
            14: std::panic::catch_unwind
                       at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/panic.rs:142:14
            15: std::rt::lang_start_internal
                       at /rustc/07dca489ac2d933c78d3c5158e3f43beefeb02ce/library/std/src/rt.rs:148:20
            16: _main
          
  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `test` (bin "test") due to 1 previous error
cargo build --release  177.36s user 47.49s system 331% cpu 1:07.90 total
```

Vollbrecht, 2024-05-25:
The install export scripts were broken upstream. I am confident that we can rule the problems out when https://github.com/esp-rs/embuild/pull/85 lands and we are using this embuild version in esp-idf-sys. I can already build against esp-idf master again with the changes. So when both gets updated it should work.

ivmarkov, 2024-06-23:
This is now fixed for the latest versions of 5.2. As in ESP IDFF `v5.2` is still broken, but `v5.2.1` and later do work.

closed 2024-06-23 by ivmarkov