esp-rs/embuild #84: Help with binding with C code

metehus, 2024-04-21:
I'm working on a new project with esp-rs with std and platformio. I can get it to work, but now I want to call C code and libraries from rust. I was searching about bindgen and came across this repository, that seems to have some kind of bindgen for building and binding c code to use in rust, with the esp environment in mind. I have tried somethings but haven't found a success. The project I'm working on is this one: https://github.com/metehus/espac/tree/main/firmware (binding in [lib/peripheral-bindings](https://github.com/metehus/espac/tree/main/firmware/lib/peripheral-bindings)).

Basically, I've looked on the exposed methods and tried this on the build file:
```rs
use std::env;
use std::path::PathBuf;
use bindgen::Builder;

fn main() {
    println!("cargo:rustc-link-search=../../.pio/libdeps/release");

    embuild::espidf::sysenv::output();
    let bindings = embuild::bindgen::run(
        Builder::default()
            .header("bindings.h")
    ).expect("error!");
}
```
There's a `bindings.h` file that just imports a library downloaded via platformio.
of course, haven't got any success:
<details><summary>Build error</summary>

```
   Compiling esp-idf-svc v0.48.1
error: failed to run custom build command for `peripheral-bindings v0.1.0 (/home/matehus/projects/embedded/espac/firmware/lib/peripheral-bindings)`

Caused by:
  process didn't exit successfully: `/home/matehus/projects/embedded/espac/firmware/target/debug/build/peripheral-bindings-c68dc13ea6345054/build-script-build` (exit status: 101)
  --- stdout
  cargo:rustc-link-search=../../.pio/libdeps/release
  cargo:rustc-env=EMBUILD_GENERATED_BINDINGS_FILE=/home/matehus/projects/embedded/espac/firmware/target/xtensa-esp32s3-espidf/debug/build/peripheral-bindings-a9cd2eac59193df5/out/bindings.rs

  --- stderr
  Output: "/home/matehus/projects/embedded/espac/firmware/target/xtensa-esp32s3-espidf/debug/build/peripheral-bindings-a9cd2eac59193df5/out/bindings.rs"
  Bindgen builder flags: ["bindings.h", "--rust-target", "1.64", "--no-derive-default", "--generate", "functions,types,vars,methods,constructors,destructors", "--"]
  thread 'main' panicked at lib/peripheral-bindings/build.rs:12:7:
  error!: Failed to generate bindings

  Stack backtrace:
     0: anyhow::error::<impl anyhow::Error>::msg
               at /home/matehus/.cargo/registry/src/index.crates.io-6f17d22bba15001f/anyhow-1.0.82/src/error.rs:83:36
     1: embuild::bindgen::run_for_file::{{closure}}
               at /home/matehus/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embuild-0.31.4/src/bindgen.rs:207:22
     2: core::result::Result<T,E>::map_err
               at /home/rust/rust/library/core/src/result.rs:829:27
     3: embuild::bindgen::run_for_file
               at /home/matehus/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embuild-0.31.4/src/bindgen.rs:205:20
     4: embuild::bindgen::run
               at /home/matehus/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embuild-0.31.4/src/bindgen.rs:170:5
     5: build_script_build::main
               at ./build.rs:9:20
     6: core::ops::function::FnOnce::call_once
               at /home/rust/rust/library/core/src/ops/function.rs:250:5
     7: std::sys_common::backtrace::__rust_begin_short_backtrace
               at /home/rust/rust/library/std/src/sys_common/backtrace.rs:155:18
     8: std::rt::lang_start::{{closure}}
               at /home/rust/rust/library/std/src/rt.rs:166:18
     9: std::rt::lang_start_internal
    10: std::rt::lang_start
               at /home/rust/rust/library/std/src/rt.rs:165:17
    11: main
    12: <unknown>
    13: __libc_start_main
    14: _start
  stack backtrace:
     0: rust_begin_unwind
     1: core::panicking::panic_fmt
     2: core::result::unwrap_failed
     3: core::result::Result<T,E>::expect
               at /home/rust/rust/library/core/src/result.rs:1030:23
     4: build_script_build::main
               at ./build.rs:9:20
     5: core::ops::function::FnOnce::call_once
               at /home/rust/rust/library/core/src/ops/function.rs:250:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

</details> 

At this point I have no idea how to make bindgen work with the platformio library and how to correctly set up the embuild (or even if that's the right tool for the job). Any information would help here!

Vollbrecht, 2024-04-21:
You may want to look in the esp-idf-sys repo and its build scripts. While its main purpose is to build esp-idf application + generating its bindings, it also has a section of the pio [implementation](https://github.com/esp-rs/esp-idf-sys/blob/master/build/pio.rs).

 Though we manly focus on the esp-idf implementation and i don't know the working status of the pio impl.

metehus, 2024-04-21:
Tried to look at it but it's way out of my knowledge. Unfortunately I will have to give up rust for this project now, but I will keep a look on the progress of this topic. I've also created an issue for better documenting this: https://github.com/esp-rs/std-training/issues/251

closed 2024-04-29 by metehus