esp-rs/embuild #87: Fails to build when platform-specific overrides are necessary in tools.json

tomstokes, 2024-06-14:
The custom tools parsing code from https://github.com/esp-rs/embuild/pull/85 does not process the overrides from tools.json.

This breaks the build on macOS and likely several other platforms. PR incoming to fix it.

Vollbrecht, 2024-06-15:
Thanks for the PR's i will review them tomorrow! Its sleep time for me now.

 Only one heads up regarding the  matching off the ARCH. The linked rust documentation at that point doesn't help because it only shows some possible values. Still possible wrong the way it is currently. I think rust itself takes the value from the target-description, and if we check at least the [supported platforms](https://doc.rust-lang.org/nightly/rustc/platform-support.html) or better directly check in the[ rustc source](https://github.com/rust-lang/rust/tree/master/compiler/rustc_target/src/spec/targets), we should check the arch value inside the target description.

If my memory is not totally wrong you can compile against rpi3 for example with armv7_unknown_linux_muslabi as well as armv7_unknown_linux_gnuabi(hf). In that cases it really seams to only use "arm" as the ARCH value. Question is if that is consistent for the rest of the target in that list.

tomstokes, 2024-06-15:
> Question is if that is consistent for the rest of the target in that list.

That's a good question. I spot checked a few of the rust compiler Target definitions before I submitted the PR, but I didn't check all of them. The ones I saw used `arm`, but of course that doesn't mean there's not a different one hidden in there somewhere.

Vollbrecht, 2024-06-15:
closed with #88

closed 2024-06-15 by Vollbrecht