esp-rs/embuild #94: [Error 267] (Windows) The directory name is invalid. If 'cargo.target-dir' contains '../'

alt-net, 2024-10-20:
Hello everyone!
I overridden taget-dir for my project, so “.cargo/config.toml” looks like this:
```
[build]
target = "xtensa-esp32-espidf"
target-dir = "..\\arcnet-target" (the same for "../arcnet-target")
```

An error occurs during the build:

```
error: linking with `ldproxy` failed: exit code: 1
  |
  = note: "ldproxy" "C:\\Users\\progm\\AppData\\Local\\Temp\\rustcJRUHiE\\symbols.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.1gsq0nmuqvxmo0590a0dd24o7.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.1hvkg5gb5v1prvv8w1e3noq7g.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.1ydkfer20rs3methae9o4npmt.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.2g7rcu064pj1y9jtx8uij6qm8.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.2zcqemf28e65plt4obc83rlmg.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.33i9hf1w1lte8h92o71n8nxbx.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.3emg1r7rp0vewtg6kbchnbdme.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.3hw7bvu5kq0co2mkybrcga54y.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.3rhy8x1857fank9hpkxcbb7fu.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.48umk6hizcee8hh4286574q7a.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.4dfpgu4t41c4locwrb08ujody.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.50ah3tjhhpv7vgoy93ommtms2.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.7b04gwc79idorvkxv7257pl15.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.9fvo21p0hwxru0d4bhn2njria.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.a69yutgtt7hl6nbnjxl4ffa2q.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.ags23s87wlt2tcomeqjiqmq03.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.asceqatbl7jvab0g64ycy2g73.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.b5lzemwt5jlasblxgp08mh6sa.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.b6bnygrivzeukh7zgzxr3n8qm.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.bmqddq9t3ia8jhfdzepbwua0f.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.c0wp70om03hlxfe8hshy6jm0y.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.c6tkit72ek2pa2gtggm7tfz1g.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.ceqppcst6tp60f3u96gtevvgv.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.ck5609z1y0zx3h795zf5lqy1u.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.di1067jooqovoctvwx9c0qmcl.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.dl4j3o0ga92uj63jgcqpva47a.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.eh1id9m2i0s7tvf0tv083cb3o.rcgu.o" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\arcnet-704b7b0923288858.5i15vnfoqjfketp1ifubj222x.rcgu.o" "-Wl,--as-needed" "-L" "V:\\arcnet\\..\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps" "-L" "V:\\arcnet\\..\\arcnet-target\\debug\\deps" "-L" "C:\\Users\\progm\\.rustup\\toolchains\\esp\\lib\\rustlib\\xtensa-esp32-espidf\\lib" "-Wl,-Bstatic" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libesp_idf_svc-c17bd6374a7d1d08.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libembassy_futures-67368fa8e808bf3e.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libembassy_time_driver-60912522c2a75e19.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libuncased-4e3774f90d2a7f02.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libesp_idf_hal-114189e234a55e20.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libembedded_hal_async-6c08ce6faf5a0d0e.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libembedded_can-1ce7d77ed4c34639.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libembedded_hal-dcdd2b9f902b4500.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libvoid-7b297b0ce714160b.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libnb-1787990c242ed5b1.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libembedded_hal_nb-e3d3d90937c04205.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libnb-01906f1da4851e02.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libatomic_waker-a04962e8255893c5.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libembassy_sync-1334e8d9d5dc852c.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libfutures_util-d1148843944b2e3a.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libmemchr-f5f964aa8edab6bc.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libfutures_io-35ed5c9f976bc125.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libslab-739cc12a000d8f33.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libfutures_channel-8c9d4a989f928803.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libpin_project_lite-437085dc7260445b.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libfutures_sink-f0f1801f99d76ecb.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libfutures_task-0f1135dbe3ec4faa.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libpin_utils-d6d7d9514fdf5c86.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libfutures_core-5564e3d6ecea0dff.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libcritical_section-20f8fbfd6a936baa.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libembedded_hal-4bea8cae79e0ca76.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libembedded_svc-5c946fb27fb8b357.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libstrum-939071bdba243b52.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libheapless-6362a0efbfdd2fc5.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libhash32-79b1af7e7ab3fb96.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libbyteorder-5cddb700da764a15.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libnum_enum-b53bc2f33adcbbe5.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libenumset-4d7ad0ec8b416d14.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libserde-37e049d21e75aa32.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libembedded_io_async-a916af954e24a082.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libembedded_io-a575a2043e952632.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libanyhow-9c099dd8528e13ae.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libesp_idf_sys-8f7ced6a7ef5a34a.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\liblibc-d08f3fae1708a55e.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libconst_format-1eabd4f9d56a3c50.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\liblog-4b8ba4cdbf795be3.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libstd-11ac8fdc13817de2.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libpanic_abort-485991982294af2b.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libobject-8ab3494fa118b28a.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libmemchr-fa999d36e0a700aa.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libaddr2line-9326f74155eb318c.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\libgimli-aa1380c2557fcb38.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\\debug\\deps\\librustc_demangle-05cd58a41079d615.rlib" "V:\\arcnet-target\\xtensa-esp32-espidf\ [... cut ...]
          Error: The directory name is invalid. (os error 267)
          
          Stack backtrace:
             0: <unknown>
             1: <unknown>
             2: <unknown>
             3: <unknown>
             4: <unknown>
             5: <unknown>
             6: <unknown>
             7: <unknown>
             8: <unknown>
             9: <unknown>
            10: <unknown>
            11: BaseThreadInitThunk
            12: RtlUserThreadStart
```

**Is there any way to fix it for Windows?**

ivmarkov, 2024-10-20:
I wonder what is this V drive:
- Option A - a real drive
- Option B - a virtual "subst" drive hiding a long directory path that the build clearly warns you should not do
?

alt-net, 2024-10-20:
V is just real hard drive partition
[image]

ivmarkov, 2024-10-20:
I have no ideas then. Maybe you can play a bit with e.g. absolute path for your target dir to figure out what is the actual problem in the first place. I.e. us it the relative path, or something else.